rsta2/uspi #3: # key

Avoncliff, 2015-03-01:
Hi
I have installed you latest version on my forth, and it runs fine. No chance to check the timings yet.
Small problem on both versions, the # key on my keyboard is not working.
It gives the same output as the |\ key, which is a bit odd. Is there a quick easy way to check the raw keys? Also on your UK map you do not have the £ (UK pound) symbol on shift 3, or the euro symbol on alt 4.
David

rsta2, 2015-03-02:
The UK keyboard map was only a best guess because I do not have a UK keyboard. Maybe the '#' is not in the right place. The keymap is in the file *lib/keymap_uk.h*. Perhaps you can correct it on your own with your keyboard and send me the correction. Thank you!

The UK Pound and the Euro symbol are not in the character set of the USPi environment as are most country specific characters. They cannot be displayed on screen because the font has only the standard ASCII 7 bit characters. So they cannot be entered on the keyboard too.

The raw keys can be got be installing a raw key status handler by *USPiKeyboardRegisterKeyStatusHandlerRaw()* (see *uspi.h*).

Avoncliff, 2015-03-02:
Hi
Thanks for the comments, well done on the UK keyboard if you do not have 
one!
I have tried the raw input and find that the #~ key is 0x31 on this 
keyboard.
I think that is standard for UK, but the other more expensive keyboard I 
have does not work with uspi, although it works on raspberian, so I can 
not confirm.
The very cheap keyboard I also have that will not work on raspberian, 
but does work on uspi, has 2 \| keys that both give code 0x31 so not 
compatible. I think this maybe US layout.
I have also put # as shift 3, this is US layout, but is a second best to £.
File attached - if it gets through....
David

rsta2, 2015-03-03:
Thanks for your response. I will change the key 0x31 to #~ in the UK keymap. Key 0x32 will stay the same. (?)

Unfortunately your attached file cannot be accessed on GitHub. Could you please send it to my Email address (noted on top of most USPi source code files). I would include the file as new US keymap then.

Not good that your more expensive keyboard is not working with USPi. Are there any messages from USPi with it? But unfortunately it is often a long time process to sort this out.

closed 2015-03-03 by Avoncliff