rsta2/uspi #9: Trouble porting to rtems

ChaosSeeker, 2016-08-09:
Hey there!

First of all great work with everything you've done. It must have taken a lot of work and it's pretty complex.

Now, I'm really new to systems programming so bear with me please. I'm trying to port your driver to work with RTEMS (a real-time operating system) on a RPI2, since I didn't find a network driver for RPI2 on RTEMS.

I've successfully ported most of it, and it's almost functional (I'm only interested in the networking side). I am now having trouble when running the samples, mainly the ethernet one, on the use of the function USPiReceiveFrame, more specifically DWHCIDeviceSubmitBlockingRequest. This is giving me a request length of 0, when it should be above 4 (according to your comments).

It's my understanding that you didn't test it on RPI2, so it can be something to do with the differences between the 2 versions? I was able to dig and find that the request being made is a bulk one and that no bytes are being transmitted since the bytes left it's still at 1600.

I'm not sure it has anything to do with it, but I'm curious, where did you find the addresses for the  core and host registers of the dwhci device? Or did I misinterpreted it and it's your own implementation all alone?

I'm running it on RPI2, u-booting with tftp. I am inclined to believe running it in RTEMS is not the problem, but I may be wrong.

If you have any ideas of what it might be or if I can provide any more info to help debug this problem please do tell me so.

Thanks for your time,
ChaosSeeker

ChaosSeeker, 2016-08-09:
Hey again. After further analysis and comparison to other drivers I noticed your comment on this line "if (nResultLength < 4)				// should not happen with HW_CFG_BIR set". I then noticed you didn't have the HW_CFG_BIR flag set, so after adding "|| !SMSC951xDeviceWriteReg (pThis, HW_CFG, HW_CFG_BIR))" to the SMSC951xDeviceConfigure function it worked out!

closed 2016-08-09 by ChaosSeeker

rsta2, 2016-08-10:
Thanks!

USPi had been ported to the RPi 2 and was tested there. My sources of information for implementing USPi are listed in *lib/README*. I don't think there is an official documentation of the USB controller available for the public.

Please note that zero is a valid return value of USPiReceiveFrame() as noted in *include/uspi.h*. This function returns zero when there is currently not a received frame available for processing. Normally you have to call USPiReceiveFrame() continuously until a value of unequal zero is returned which means a frame was received. After processing this frame you should continue to call USPiReceiveFrame(). This takes some CPU time but there is no other way to continuously poll the Ethernet device and to allow concurrent USB transfers (e.g. for accessing an USB keyboard) at the same time.

Setting HW_CFG_BIR causes the Ethernet device to postpone the acknowledgement of the USB transfer until a received frame is available. If you do this USPiReceiveFrame() should not return zero any more but you cannot use an USB keyboard, mouse or gamepad any more at the same time. Because USPi by default should be able to use these devices at the same time I cannot set HW_CFG_BIR. But if you want to use the Ethernet device only it may work successfully with HW_CFG_BIR in your scenario.

ChaosSeeker, 2016-08-10:
Thanks so much for your response! I hadn't understood exactly how the flag
worked or why the result would be zero. Now I get it and can continue to
build upon it!

Since I'm only interested in the Ethernet device I will probably leave
HW_CFG_BIR on.

Thanks again for the quick response!