rsta2/uspi #10: USB serial... is it possible?

fbergama, 2016-09-14:
Since some raspberry pis have USB OTG built in (like the zero), how difficult would it be to implement USB serial so that the Pi can act as a serial device when plugged to a pc?

Alternatively, it could be enough for me to just communicate with the pi via libusb (I mean I don't care for the full USB serial emulation). 

Thank you and cheers for your excellent project.

rsta2, 2016-09-15:
Thanks for your interest in USPi and for your words about it.

Before developing USPi I analyzed the Linux DWG OTG driver but to reduce complexity I ignored everything which had to do with USB device mode which was a lot. So USPi is an USB host mode only driver.

Because of this it would be a totally new project to support device mode with or without serial support. Unfortunately I'm currently not able to do this work.

fbergama, 2016-09-15:
I was afraid that the "device mode" implies actually a completely different behavior of the driver/APIs. So, I agree that it should be a separate project.

I have some prior expertise in bare metal raspi programming (see my [PiGFX](https://github.com/fbergama/pigfx) project) but I have absolutely no idea of the inner working of a USB software stack. Supposing that I want to implement the minimum USB device mode functionality so that i can send/receive packets from a host pc, how much of your USPI code can be recycled for that (like, don't know, peripherals enumeration, crc check, etc)? Do you think that the Linux DWG OTG driver you mentioned would be sufficient to start with?
In case, would you like to participate with me (maybe just by sharing your expertise in the area) in a project like that? Of course, just as a low-priority hobby project...

rsta2, 2016-09-16:
I don't think the USPi code is of much use in USB device mode because we are on the "opposite" side of the USB here. This is very different to host mode. E.g. the device has to respond to enumeration while the host controller driver initiates device enumeration. CRC check is done by the USB controller itself. There's no need to handle this in software.

I think the Linux DWC OTG driver (DWG OTG was a typo!) is the only existing public reference. So you would have to start with it. There is much "research" work to be done on it to understand how device mode is working. You may also have a look at [this forum topic](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=129653) and the [USB 2.0 specification](http://www.usb.org/developers/docs/usb20_docs/).

If you have specific questions you can write an Email to me (see my address in most USPi source code files).

fbergama, 2016-09-16:
I had a quick look at the DWC OTG driver and I think that it would be a huge endeavor for me to implement or simply adapt something like that. I don't have enough time to gain the needed expertise on the USB specifications  to even understand how the code is supposed to work.

As a future reference, I think that the latest version of U-boot implements such feature. Probably it would be easier to start from their code instead of the Linux driver you mentioned.

Thank again for your support!

closed 2016-09-16 by fbergama