rsta2/uspi #12: Reading and writing data to usb storage

Prathik4, 2017-02-02:
Hello,
First of all you had done some great work. I'm new to the embedded world and really trying to keep up and learn things and your examples have helped a lot.
I just want to know if it is possible to read and write data to/from usb file system. I'm working on a academic project where I need to store some output while running some control's code on raspberry pi 2. In C++ bare metal environment (circle), example number 15 do store some data to usb filesystem in .txt file after initializing the usb drive. So i was wondering if it was possible with the uspi bare metal environment too.
Thanks in advance.

rsta2, 2017-02-02:
Unfortunately USPi does not contain a file system driver. It does only allow reading/writing physical sectors from/to an attached USB disk. If you need file access to a FAT partition you have to use Circle.

Prathik4, 2017-02-02:
Thank you for the replying back. 👍

closed 2017-02-03 by Prathik4

samimash, 2017-06-21:
> Unfortunately USPi does not contain a file system driver. It does only allow reading/writing physical sectors from/to an attached USB disk. If you need file access to a FAT partition you have to use Circle

i understand what reading sectors from an attached usb mean,but what do you mean by writing?