rsta2/uspi #13: using the library

samimash, 2017-06-13:
i am stuck [here](https://github.com/rsta2/uspi#using)

> Add the USPi include/ directory to the include path in your Makefile and specify the libuspi.a library to ld. Include the file uspi.h where you want to access USPi functions.
 
i found a file called makefile in the lib directory 
what should i do with it?
  
[image]

rsta2, 2017-06-14:
You have to do nothing with this Makefile. It is internal to USPi. If you want to add the USPi library to your own project, you will have to create your own project Makefile. There should be some include path definition in your Makefile where you have to add the USPi include/ directory. Further you have to add the libuspi.a file (with path) as a library for the linker (ld), where it is called in your Makefile.

samimash, 2017-06-15:
a link to a tutorial for this will be useful

rsta2, 2017-06-16:
Perhaps it's the easiest if you try to modify one of the USPi samples for your purpose. But if you need special features, you also have to modify the USPi Environment, wish may be not easy. USPi requires some programming knowledge to be used. I cannot tell you a link to a (make) tutorial, because I do not know one.

JonnyLatte, 2017-06-19:
sammimash: this tutorial for a different pi bare metal project might help you 

http://www.valvers.com/open-software/raspberry-pi/step01-bare-metal-programming-in-cpt1/

juanitoddd, 2017-06-22:
Hello, 
I am also kind of stuck here,
I am attempting to use a MIDI-USB cable to read Notes on the RPI3:
After building a Sample (MIDI) using the Makefile, I got this new files 
kernel.elf  kernel.img  kernel.lst  kernel.map
But I dont see any Excecutable to test it out.  How can I achieve this ?
Thanks

rsta2, 2017-06-23:
*kernel.img* is the executable image. You have to put it on a SD card together with the firmware files *bootcode.bin*, *start.elf* and *fixup.dat*. These files can be downloaded [here](https://github.com/raspberrypi/firmware/tree/master/boot). Put the SD card into your Raspberry Pi and start it.

closed 2017-08-20 by rsta2