rsta2/uspi #15: USB difference between Pi 1, Pi 3?

7043mcgeep, 2018-06-08:
### Preface: If you are unable to answer my following question, how can I get some kind of logging output with USPi (e.g to a readable file), so I can view the order and details of the device setup?

Thank you for making this implementation accessible. I was able to build and run on my Pi 3 B+, yielding a solid LED light on its Ethernet port.

I have a USB driver that works on the Pi 1 boards. But it _fails_ on the Pi 3 and 3 B+ boards. I was told the peripherals of the BCM2835 are the same as the 2837 and 2837B0. However, my driver's root hub abstraction (which I believe is the equivalent of USPi's "standard hub") fails to attach to the DesignWare Hi-Speed USB 2.0 On-the-Go Controller, because it fails to read its device descriptor.

Is there any major difference between the Pi 1 and Pi 3's USB architecture that I am not understanding?

Here is a log of the failed attachment I'm getting. There are 8 channels (0-7) and the transfer seems to always occur on 7. I have tried hard-coding for channel 6, but that fails, too. I think my problem may be at line 17, as I am cautious to write it off as an issue of signedness, since it's so close to the error.

Let me know if you notice anything funky, or have any insight. And thanks.

```
[1]   Transfer begins after transfer request sent
[2]   **SETUP** packet, direction = OUT
[3]   SOF interrupt, channel halts (signaling transfer end) (interrupts pending: 0x00000023
[4]   1 packet transferred on channel 7, 8 bytes transferred
[5]   **DATA** packet, endpoint direction = IN
[6]   SOF interrupt, channel halts (signaling transfer end) (interrupts pending: 0x00000023
[7]   1 packet transferred on channel 7, 8 bytes transferred
[8]   **STATUS** packet, endpoint direction = OUT
[9]   SOF interrupt, channel halts (signaling transfer end) (interrupts pending: 0x00000023
[10] 1 packet transferred on channel 7, 0 bytes transferred
[11] **Assigning address 2 to new device**
[12] **SETUP** packet, endpoint direction = OUT
[13] SOF interrupt, channel halts (signaling transfer end) (interrupts pending: 0x00000023
[14] 1 packet transferred on channel 7, 8 bytes transferred
[15] **STATUS** packet, endpoint direction = IN
[16] SOF interrupt, channel halts (signaling transfer end)
[17] 1 packet transferred on channel 7, 4294443016 bytes transferred (Actual transfer size 0 of 0 bytes)
[18] **SETUP** packet, endpoint direction = OUT
[19] SOF interrupt, channel halts (signaling transfer end) (interrupts pending: 0x00000082)
[20] Device 2: transfer **error** on channel 7
[21] Device 2: **Failed** to read device descriptor: hardware error
[22] Device 1: **Failed** to attach new device to port 1: hardware error
```

7043mcgeep, 2018-06-10:
Thanks for your reply. Can you tell me the initialization process of your implementation? I'd like a verbose explanation of the attachment process from boot to device attachment, but whatever length it takes you to get the point across is fine (even numbered steps).

And if you could accompany those steps with relevant filenames, that would be nice.

Aside from steps involved, the most helpful elaborations would be on packets sent during the attachment process (descriptors, status, etc).

USPi is most desirable to learn from because you list references that I am familiar with. Additionally, USPi is more readable, lighter in weight, and more updated than similar bare metal drivers.

It's good that this ported-from-C++ implementation works. It's just that the comments provide almost no explanation to the purpose of any file, let alone its odd source code. Hence, my need for further explanation.

I appreciate your reply.

rsta2, 2018-06-11:
Excuse me, but explaining the entire initialisation process in detail would be a long story and is not possible for me. You can only read the source code to get this information. I would have to do this too now, because I have written this code over 3 years ago and I am not aware of the details any more.

7043mcgeep, 2018-06-11:
>  explaining the entire initialisation process in detail would be a long story and is not possible for me. 
> You can only read the source code to get this information.

The source code is difficult to read and, as you put, _strange_. I am only interested in the events up to and including USPi's initial attachment of the on-board Ethernet controller. Nothing more. Maybe you can take a look at your repository and get back to me when you have a better idea.

I imagine this C implementation uses the same methods of attachment as your extensive Circle USB library. Thus, I assumed it wasn't a far-fetched question to ask about the code you wrote.

7043mcgeep, 2018-06-12:
An update: I found that your logger.c file provides function definitions that I can use to get potentially useful output. I'll keep this issue thread updated as I discover USPi's initialization process.

rsta2, 2018-06-21:
I think, the [Raspberry Pi bare metal forum](https://www.raspberrypi.org/forums/viewforum.php?f=72) is a better place to discuss this. So this issue can be closed.

closed 2018-06-21 by rsta2