rsta2/uspi #19: Doesn't compile with aarch64.

gnaghi, 2018-10-18:
With aarch64 GCC version, there are errors in the compilation :
- missing immediate expression at operand 1 -- `dmb' => can be corrected by changing
`__asm volatile ("dsb" ::: "memory")`
to
`__asm volatile ("dmb sy" ::: "memory")`
- unknown or missing system register name at operand 2 -- `mrs x1,cpsr'. cpsr no longer exist in arm64. PSTATE should be used instead.
- unknown mnemonic `cpsid' -- `cpsid i' => can be corrected by changing to `msr daifclr, #2`
- unknown mnemonic `cpsie' -- `cpsie i' =>_can be corrected by changing to `msr daifset, #2`
- unknown mnemonic `mcr' -- `mcr p15,0,x0,c7,c14,1'

I didn't found yet a solution for this one, but it looks like mcr should be changed to msr (https://stackoverflow.com/questions/32374599/mcr-and-mrc-does-not-exist-on-aarch64)

rsta2, 2018-10-18:
The USPi library is currently not intended to be used with AArch64. Porting it to this architecture requires probably some additional work (compared to the issues you have noted), I haven't done yet, because this is the first time, that there is demand for it. I will have a look on it, when I'm ready with my current project, probably in November.

gnaghi, 2018-10-19:
No problem. It's more a report for those who would try the experiment :)

rsta2, 2018-10-19:
OK. I will have a look on it in spite of it then.

rsta2, 2018-10-24:
The USPi library can be built with AArch64 support now. Please have a look at the end of the main *README.md* file for further info.

AArch64 support in USPi does not come with an environment library. You need your own OS environment to use it. For testing I used the [Circle64](https://github.com/rsta2/circle64) environment. I only tested the USB keyboard function so far with the attached sample program. The sample programs which are provided with USPi cannot be used at the moment with AArch64.

[uspitest64.zip](https://github.com/rsta2/uspi/files/2510850/uspitest64.zip)

gnaghi, 2018-11-22:
Great, thank you, I'll test it ASAP :)

winstyfreddy, 2018-12-20:
Hi René,

Many thanks for this. I've successfully built the test kernel from [uspitest64.zip](https://github.com/rsta2/uspi/files/2510850/uspitest64.zip), but I'm getting the rainbow splash screen when booting into the new kernel. Is a special `config.txt` file needed? I did as follows:

1) Install the [gcc 8.2 toolchain from the ARM Developer portal](https://developer.arm.com/-/media/Files/downloads/gnu-a/8.2-2018.11/gcc-arm-8.2-2018.11-x86_64-aarch64-linux-gnu.tar.xz?revision=7a60a425-1aa0-43f5-b9db-1af71bffadc6&la=en).

2) Set `PATH` to use it.

3) Clone circle64 repo (revision [`f1f440fa`](https://github.com/rsta2/circle64/commit/f1f440fac3099db137b70e0f26f73f2240720db1)) in home folder

4) Clone uspi repo (revision [`01fb7b49`](https://github.com/rsta2/circle64/commit/01fb7b49f83f1750dee2044dc923b11d5d775115)) in home folder

5) Download and extract [uspitest64.zip](https://github.com/rsta2/uspi/files/2510850/uspitest64.zip) in home folder

6) Apply the following changes:

```diff
pmoore@ubuntu:~/circle64$ git diff
diff --git a/Rules.mk b/Rules.mk
index d7316a7..0702e46 100644
--- a/Rules.mk
+++ b/Rules.mk
@@ -24,7 +24,7 @@ endif
 
 -include $(CIRCLEHOME)/Config.mk
 
-RASPPI ?= 3
+RASPPI = 3
 PREFIX ?= aarch64-linux-gnu-
 
 CC     = $(PREFIX)gcc
```

```diff
pmoore@ubuntu:~/uspi$ git diff
diff --git a/Rules.mk b/Rules.mk
index daa4454..00695bd 100644
--- a/Rules.mk
+++ b/Rules.mk
@@ -24,7 +24,8 @@ endif
 
 -include $(USPIHOME)/Config.mk
 
-AARCH64        ?= 0
+# AARCH64      ?= 0
+AARCH64        = 1
 
 ifeq ($(strip $(AARCH64)),0)
 RASPPI ?= 1
```

7) Update `CIRCLEHOME` and `USPIHOME` in uspitest `Makefile`:

```
pmoore@ubuntu:~/uspitest$ cat Makefile 
#
# Makefile
#

# CIRCLEHOME = ../..
# USPIHOME   = ../../../uspi

CIRCLEHOME = ../circle64
USPIHOME   = ../uspi

CFLAGS	+= -I $(USPIHOME)/include -DAARCH64=1

OBJS	= main.o kernel.o uspibind.o

LIBS	= $(USPIHOME)/lib/libuspi.a \
[... 119 lines cut ...]
aarch64-linux-gnu-g++ -I ../uspi/include -DAARCH64=1 -march=armv8-a -mtune=cortex-a53 -mlittle-endian -mcmodel=small -Wall -fno-builtin -nostdinc -nostdlib -D__circle__ -DRASPPI=3 -I ../circle64/include -O2  -fno-exceptions -fno-rtti -std=c++0x -c -o main.o main.cpp
aarch64-linux-gnu-g++ -I ../uspi/include -DAARCH64=1 -march=armv8-a -mtune=cortex-a53 -mlittle-endian -mcmodel=small -Wall -fno-builtin -nostdinc -nostdlib -D__circle__ -DRASPPI=3 -I ../circle64/include -O2  -fno-exceptions -fno-rtti -std=c++0x -c -o kernel.o kernel.cpp
aarch64-linux-gnu-g++ -I ../uspi/include -DAARCH64=1 -march=armv8-a -mtune=cortex-a53 -mlittle-endian -mcmodel=small -Wall -fno-builtin -nostdinc -nostdlib -D__circle__ -DRASPPI=3 -I ../circle64/include -O2  -fno-exceptions -fno-rtti -std=c++0x -c -o uspibind.o uspibind.cpp
aarch64-linux-gnu-ld -o kernel8.elf -Map kernel8.map -T ../circle64/circle.ld ../circle64/lib/startup.o main.o kernel.o uspibind.o ../uspi/lib/libuspi.a ../circle64/lib/libcircle.a
aarch64-linux-gnu-objdump -d kernel8.elf | aarch64-linux-gnu-c++filt > kernel8.lst
aarch64-linux-gnu-objcopy kernel8.elf -O binary kernel8.img
wc -c kernel8.img
148944 kernel8.img
pmoore@ubuntu:~/uspitest$ 
```

The resulting md5 shas I got were:

```
$ pmoore@ubuntu:~/uspi64$ md5sum *
8e17173bee9a0da151e65c235cc78f4d  bootcode.bin
b616a9c14d8ee516e630ebe04ab34946  fixup.dat
55d31b8730a865e9d00293f70557dcef  kernel8.img
4a4d169737c0786fb9482bb6d30401d1  LICENCE.broadcom
26f983401f6473eaa55018b9dca2bd68  start.elf
```

I have a [serial device](https://www.amazon.de/gp/product/B072K3Z3TL) that I can connect if there is debug output I can grab, and I can also supply you with the generated build files if you like. I'm using a Raspberry Pi 3B+.

Many thanks!

winstyfreddy, 2018-12-20:
Ah, re-reading the docs, maybe I need to customise [this code](https://github.com/rsta2/uspi/blob/38eaff4f715643a9a5a7038470b14ac351861475/include/uspios.h#L29-L48) first?

If that is the case, can you share with me the values you use? I have both a US keyboard and a DE keyboard. Note, with the [32 bit USB keyboard demo](https://github.com/rsta2/circle/tree/master/sample/08-usbkeyboard) only my DE keyboard works, so that might be the best one to build for.

rsta2, 2018-12-21:
Hi Winston,

this is probably an toolchain incompatibility issue. Please download and use [this toolchain](https://developer.arm.com/-/media/Files/downloads/gnu-a/8.2-2018.11/gcc-arm-8.2-2018.11-x86_64-aarch64-elf.tar.xz?revision=90d1fe0a-42d7-439c-8ea6-bb7234b2f540&la=en) from the ARM Developer website and try again. It's the AArch64 ELF bare-metal target (aarch64-elf). It tried it and it is working here. You have to set `PREFIX = aarch64-elf-` in Config.mk or Rules.mk to use it. I was using the GCC 7.2.1 toolchain from Linaro before and the aarch64-linux-gnu target was working there too.

A config.txt file is not needed. The keyboard configuration should be set before build. But this effects only the keyboard map. If the keyboard is not found with a different language setting, this would be another issue.

rsta2, 2018-12-21:
I have to clarify one thing. Toolchain version are important and not any toolchain will work. To handle this, there is or will be a recommended toolchain for my projects. That is the GNU C 7.3.1 based toolchain referenced in the main *README.md* file of the respective project at the moment. I have updated the [Circle64](https://github.com/rsta2/circle64) project this way.

winstyfreddy, 2018-12-27:
Many thanks @rsta2!

I'm happy to say this is working for me now. For anyone else that stumbles on this issue, here is the script I used:

```
#!/bin/bash -xve

NEW_DIR=$(mktemp -d)
cd "${NEW_DIR}"
curl -sL 'https://releases.linaro.org/components/toolchain/binaries/latest-7/aarch64-elf/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-elf.tar.xz' | tar xvfJ -
export PATH="$(pwd)/gcc-arm-8.2-2018.11-x86_64-aarch64-elf/bin:${PATH}"

git clone git@github.com:rsta2/circle64.git
cd circle64
./makeall clean
./makeall
cd boot
make
cd ../..

git clone git@github.com:rsta2/uspi.git
cd uspi
{
  echo 'AARCH64 = 1'
  echo 'RASPPI = 3'
  echo 'PREFIX = aarch64-elf-'
  echo "USPIHOME = $(pwd)"
} > Config.mk
cd lib
# No `make clean`, so skip it
# make clean
make
cd ../..

curl -sL 'https://github.com/rsta2/uspi/files/2510850/uspitest64.zip' > uspitest64.zip
unzip uspitest64.zip
cd uspitest
sed -i "s%CIRCLEHOME = .*%CIRCLEHOME = ${NEW_DIR}/circle64%" Makefile
sed -i "s%USPIHOME   = .*%USPIHOME   = ${NEW_DIR}/uspi%" Makefile
make
cd ..
mkdir sdcard
mv uspitest/kernel8.img sdcard
mv circle64/boot/* sdcard
md5sum sdcard/*

echo "Files in ${NEW_DIR}/sdcard"
```

@rsta2 one small thing - the red text spinning character spins tremendously fast, in the top-right of the screen. It looks like there is no pause between each drawing of the character - could this be a problem with reading one of the clock signals? I can make a video if you like.

For me this issue is closed now, as the above script builds the `uspitest` program for me successfully.

I won't close it though, as it was created by @gnaghi.

Many thanks!

rsta2, 2018-12-28:
@winstyfreddy It's good, that it is working now for you. Thanks for providing your script!

I think, there is no delay in the main loop of the sample program. So the "rotor" spins as fast as it cans. I don't think, there is a problem with the clock.

I will wait some days and will close this issue then.

closed 2019-01-02 by rsta2