rsta2/uspi #20: Unplugging USB keyboard causes error messages

winstyfreddy, 2018-12-28:
When unplugging USB keyboard the following error message is continuously logged:

```
dwhci: Transaction failed (status 0x82)
```

After reattaching USB keyboard, the message continues to be logged.

winstyfreddy, 2018-12-28:
I should add, this is when using `uspi` in combination with `circle64` environment - I'm not sure if this is a `circle64` issue or a `uspi` issue, but logging it here first.

Also note, there is no error if the keyboard is first attached after the kernel has booted. The kernel correctly reports `kernel: Keyboard not found`. However, if the keyboard is attached afterwards, it is not detected and cannot be used. Again, not sure if that is a `circle64` issue or `uspi` issue.

Many thanks René.

winstyfreddy, 2018-12-28:
> Also note, there is no error if the keyboard is first attached after the kernel has booted. The kernel correctly reports `kernel: Keyboard not found`. However, if the keyboard is attached [...]

On closer inspection, I see the `uspitest` program is what logs that the keyboard is not found, and aborts - so that isn't an issue with `uspi` or `circle64`, after all. Apologies.

So I think the only issue is the one from the first comment, that if the keyboard is unplugged, `dwhci` logs this error repeatedly.

rsta2, 2018-12-28:
The USPi library does not support hot-plugging. USB devices must be connected before boot and must not be removed as long the system runs.

winstyfreddy, 2018-12-28:
Many thanks @rsta2, no worries. Thanks for the quick response!

I'll close this then. I've been collecting a list of resources to work my way through the process of understanding USB on the RPi [here](https://github.com/spectrum4/spectrum4/blob/master/USB/README.md), let me know if you think I've missed any useful links!

And again, many thanks for this great library. 👍

closed 2018-12-28 by winstyfreddy