rsta2/uspi #21: RPi 4 support ?

penfold42, 2019-06-24:
How tricky will this be ?

rsta2, 2019-06-25:
This requires a complete rewrite of the USB host controller driver. It is not likely, that there will be a version of this project, which can be used with the RPi 4, in foreseeable time.

penfold42, 2020-05-10:
Now that circle has support for Pi4 is uspi support more probable ?

Thanks !

rsta2, 2020-05-11:
The effort to port this from C++ to C remains huge, I'm sorry. It's not only the xHCI driver. USB on the RPi 4 needs MMU LPAE support (for AArch32) and a driver for the PCIe host bridge too. And if one wants Ethernet access, this would be another separate driver. With the increasing variety of RPi hardware, I decided to focus on Circle and related projects.

cleverca22, 2020-09-29:
the pi4 still has the old dwc2 controller as well, but its now wired to the usb-c port

the USB_OTG pin isnt connected properly, and the fancy usb-c direction detection logic is missing, so you have to force host or device mode like with the pi A's

MikeDB1, 2020-09-29:
Why not consider converting your project to Circle so you have all the features of it ?   I converted a huge Linux C 32 bit program to run on Circle 64 bit in only a few weeks with excellent results.  You don't have to use all the features of Circle and indeed I used my own audio drivers, but it's still an excellent base for bare metal work.  There isn't much it can't do.

seamusdemora, 2020-10-01:
"Circle"?? Are you referring to this? https://github.com/circleci

seamusdemora, 2020-10-01:
@MikeDB1 : Thanks! The search I did ([`github circle`](https://duckduckgo.com/?t=ffnt&q=github+circle&ia=web)) didn't reveal that. So glad to know of this!

Maybe you can answer another question for me: I'm interested in a reliable method for turning USB power ON and OFF. [This item from the Linux kernel docs](https://www.kernel.org/doc/Documentation/usb/power-management.txt) indicates it's supported in the kernel, but indicates it must also be supported by the driver. I can't find this "USB power" feature described in the RPi documentation, and so assume it's just not supported in the "native" driver. The `uhubctl` project indicates it's possible to do this on the RPi, but I guess it must be using `sysfs`? At any rate - integration with the driver would seem to be a "cleaner" way to do this. Is anybody here in circle doing something like this?

MikeDB1, 2020-10-01:
I seem to recall a comment on the forum that the only way to do this was to use a special cable, for example to connect a Pi4 to a PC, with the power lines disconnected. That would appear to suggest that the Pi driver has no method to turn power off.

seamusdemora, 2020-10-01:
Perhaps... but I wonder how [`uhubctl`](https://github.com/mvp/uhubctl) manages to do it?

MikeDB1, 2020-10-01:
Seems there was a firmware update from VLI superceding the comment I'd seen.  
https://www.raspberrypi.org/forums/viewtopic.php?t=258489
I'll have a play with it.

MikeDB1, 2023-11-09:
> I am past half way through converting the latest (2023) version of Circle to C. I am mainly focused on 4B+ support since that's where there is the least C support. Check it out here<https://macoy.me/code/macoy/rpi-system>.

Does it run any faster than the C++ version ?

makuto, 2023-11-09:
> Does it run any faster than the C++ version ?

I haven't put any time into optimizing it over where Circle is; it's close to 1:1 at this point. You can expect slightly faster compilation times thanks to better header files (and C just compiles faster anyways), but otherwise it should be the same as Circle. I mainly did it because I need everything in C, not because of performance.

rsta2, 2023-11-10:
To have an decision here: There won't be an USPi version for Raspberry Pi models 4 and up. So this issue can be closed.

Please discuss topics, which affect mainline Circle in [Circle Discussions](https://github.com/rsta2/circle/discussions). Discussions on forks should be provided by the owner of the fork. GitHub makes it easy to setup your own discussion in your project.

closed 2023-11-10 by rsta2