rsta2/uspi #23: Not working with some keyboards?

zaphodbe, 2020-02-17:
So far using the USPI library as part of PiGFX I seem to have problems receiving key strokes from a USB keyboard.  As noted in PiGFX some keyboards are not detected.  However I have several that are being detected successfully, however I never seem to get the key strokes from them.  I added some debugging to the code and it appears the keyboard handler is not getting called. 

Any ideas?

Thanks
Chris

rsta2, 2020-02-18:
It is known, that some keyboards are not working with USPi, but most should work. Have you tried the USPi keyboard sample program with your keyboards?

zaphodbe, 2020-02-18:
Yes was aware of that.  The issue turned out to be with PiGFX.  The timer implementation that is part of PiGFX wasn't working correctly when called with a delay of 0.  I posted a suggested fix there.

closed 2020-02-18 by zaphodbe