rsta2/uspi #24: Can't initialize USPi on Raspberry Pi 3B+

Leonard-Blazevic, 2020-03-03:
Hi, 

Before trying to use your library inside a project, I tried to run the ethernet sample you provided to get familiar with the code, but when I build it as is, I don't see any output and the board seems stuck.

I have tracked down the problem to this function call in the timer initialization => https://github.com/rsta2/uspi/blob/master/env/lib/timer.c#L306 which seems to never finish.

If I comment out that line, I see the following output:

[image]

After some debugging, I think it is related to the fact that the timer is not properly initialized which causes problems in the USB initialization and the DeviceNameServiceGetDevice function to never find any devices.

I added the following lines to the Rules.mk:

`AARCH64 = 0`
`RASPPI = 3`
`PREFIX = arm-none-eabi-`

Do you have an idea how could I go about fixing this issue?
Thanks,
Leonard

rsta2, 2020-03-04:
Unfortunately I cannot reproduce this issue. I built the Ethernet sample with the same settings with a GCC 9.2.1 toolchain again and tested it with a firmware from Feb 12 2020 from [here](https://github.com/raspberrypi/firmware/tree/f4b58692fef0b9c16bd4564edb980fff73a758b3/boot) and it works.

The `TimerMsDelay()` function is using the system timer and simply reads the CLO register of this device, which returns the current lower 32 bits of a free running 1 MHz timer. There is no initialization needed to start this timer.

Can you try the firmware I mentioned above? How did you debug this issue?

rsta2, 2023-11-10:
There was no further reaction of the originator of this issue for a long time. The issue will be closed, because it cannot be solved without interaction.

closed 2023-11-10 by rsta2