rsta2/uspi #25: Odd Hang waiting for DWHCI device (Pi 3B+)

Pear0, 2020-03-12:
I'm getting a very similar behavior to #22 where my Pi 3B+ hangs. I have configured the timers and interrupts. The kernel is a custom preemptive kernel. The process managing uspi is pinned to one core.

Oddly enough if I delay the initialization of the uspi driver process for one second on boot, everything initializes and works fine, without getting stuck on line 719.

Do you have any idea what could be causing this behavior? I'd rather not leave the delay in because I want to get networking online as soon as possible during boot.

rsta2, 2020-03-12:
Is the USB IRQ handled on the same core as the USPi driver process is running? If not you may have to insert a DataMemBarrier() after line 721 and a DataSyncBarrier() before line 733. The driver is waiting for completion of a synchronous transfer in line 719. If it hangs there, it won't be informed about the completion. This may have several reasons. The missing barriers may be one reason in an multi-core environment. An other reason may be, that the IRQ handler is not called.

Pear0, 2020-03-15:
I tried add memory barriers but it didn't seem to help, and it turns out that while the IRQ is being registered, I'm counting all IRQs and it never seems to get called. Is there some sort of waiting period at boot before the USB controller can be enabled and will respond?

rsta2, 2020-03-15:
I'm afraid it's difficult to use this driver in a multi-core environment. This will only work, if all execution in the driver (initialization, interrupt handler, timer handler, calls from other user processes) are run on the same core. Otherwise the simple synchronization in uspi_EnterCricital(), which simply disables the IRQ will not safely protect the critical data structures. You may have to use spin locks instead.

Pear0, 2020-04-02:
I'm pretty sure I had all the network code to only have affinity on one core, but anyway I'm going to close the issue for now because I don't have an isolated test case and it could easily be an issue with my usage.

closed 2020-04-02 by Pear0