rsta2/uspi #26: USB Timer Problem

chregu82, 2020-05-09:
Hello
I'm using the uspi library in the PiGFX project and ran into a problem.
In usbendpoint.c there is a variable called m_nInterval which is an interval in milliseconds. This value seems to be at least 1.
In function DWHCIDeviceChannelInterruptHandler this value gets converted by MSEC2HZ.

#define MSEC2HZ(msec)		((msec) * HZ / 1000)

HZ is defined as 100. If m_nInterval is less than 10 this results in a value of 0.

With this value a timer is started, which results in a division by zero in my project.

I think that this should calculate how many interrupt intervals you have to wait, before calling the function again. However, HZ should then be defined at least 1000.

I'm sorry if I didn't fully understand this.

rsta2, 2020-05-09:
The value calculated with the *MSEC2HZ* macro is handed over directly to the *StartKernelTimer()* function, which has to be provided by your bare metal environment. *StartKernelTimer()* must be able to handle a delay value of zero, which means that the timer handler should be called at the next timer tick.

chregu82, 2020-05-09:
Ok I understand. Thank you

closed 2020-05-18 by chregu82