rsta2/uspi #27: Alignment problem on RPI3 B V1.2

chregu82, 2020-05-18:
Hello
I try using uspi on a Raspberry Pi 3B V1.2. I ran into a problem with the initialisation. After printing out the MAC Address it runs into an alignment exception.
The uspi library tries to read a 2 byte value from a odd memory address.
This happens in USBEndpoint2 function on this line:
pThis->m_nMaxPacketSize = pDesc->wMaxPacketSize;
wMaxPacketSize really starts on a odd address, as does the start of the whole struct.
I'm lost in further debugging, please help.

Uspi allocates 39 bytes which start at address 540c4. pDesc starts at 540dd which is part of those 39 bytes.

chregu82, 2020-05-18:
I was able to resolve the issue by defining wMaxPacketSize as 2 unsigned chars and calculate m_nMaxPacketSize with LowerByte+HigherByte*256. This seems to work fine.

rsta2, 2020-05-19:
Normally the MMU should be on if using USPi and it should be configured so that unaligned memory accesses are allowed. The optimizer of newer GCC versions sometimes generates an unaligned access, where it wouldn't be expected from the code. I could implement this workaround, you have mentioned in your second comment, on the *develop* branch for the moment, if this helps you, but I cannot guarantee that this is the only unaligned access, which may happen.

chregu82, 2020-05-19:
Please do this, then i can include this branch in my project for now. It probably works for me. I'm using version 7.4 of the toolchain.

rsta2, 2020-05-19:
OK, I have done it. The patch is available on the *develop* branch.

chregu82, 2020-05-20:
Ok, i changed to the develop branch. After getting it compiling it got worse. It still crashes on PI2 and PI3 on this line:
pThis->m_nMaxPacketSize = pMaxPacketSize[0] | (u16) pMaxPacketSize[1] << 8;
My guess is, that the compiler still generates a 2byte access. I redeclared wMaxPacketSize as array of 2 unsigned chars. This worked.
On the Pi1 however this does not crash. But I'm getting the following error with my keyboard connected:
[ ERROR ] dwhci: Transaction failed (status 0x102)
This did not happen with the main branch.

chregu82, 2020-05-20:
UPDATE:
I have a working solution, please replace
pThis->m_nMaxPacketSize = pMaxPacketSize[0] | (u16) pMaxPacketSize[1] << 8;
with
pThis->m_nMaxPacketSize = (pMaxPacketSize[1] << 8) + pMaxPacketSize[0];

rsta2, 2020-05-20:
It's getting strange now. It is working here, built with GCC 9.2.1. I have to analyze this, when I have some more time. Please use your own solution for the moment. You could create a fork for this.

chregu82, 2020-05-20:
I changed my compiler to version 9.3.1. I have absolutely the same problem. Pi1 boots with USB errors, Pi2&Pi3 have exception.
I compared the compiler output between adding the bytes and oring them. It looks like the compiler is smarter than we are. It recognizes that it's easier to just load the whole word and does exactly that. This way it still crashes if it's not aligned. It's getting worse on the Pi1 which does not crash (I dont' know why) but calculates very strange values which make absoluteley no sense.

This:
pThis->m_nMaxPacketSize = pMaxPacketSize[0] + (pMaxPacketSize[1] << 8);
Becomes:
ldrb	r2, [r5, #5]
ldrb	r3, [r5, #4]
add	r3, r3, r2, lsl #8
str	r3, [r4, #12]
Which loads 2 bytes and adds them with one byte shifted by 8 bits.

This:
pThis->m_nMaxPacketSize = pMaxPacketSize[0] | (pMaxPacketSize[1] << 8);
Becomes:
ldrh	r3, [r5, #4]
str	r3, [r4, #12]
Which is just load a word and store it.

Would you consider to use add instead of or? Or maybe we should check why this whole struct is not always 2 byte aligned. Maybe this is a problem caused by my environment.

rsta2, 2020-05-21:
`wMaxPacketSize` is a member of an USB endpoint descriptor, which is a member of an USB interface descriptor by itself. It is normal, that `wMaxPacketSize` can be unaligned in memory. The environment must be able to access unaligned data by default. In the USPi environment, which is the basis for building the USPi samples, this is accomplished by using the MMU and initialize it in a way, that unaligned memory accesses are allowed. You should do this too in your own environment, if using USPi.

I thought, that it is possible to provide a workaround for USPi for accessing `wMaxPacketSize` in a different environment, but now I think, this is not a good idea any more. There may be other places in the code, where unaligned accesses are done and they will not work too, without a proper MMU initialization. Furthermore the ARM CPU is relatively slow without the MMU enabled, maybe to slow for using USPi in all cases.

chregu82, 2020-05-21:
I see... Eliminating this single unaligned access would have been the easiest solution for me. So I'm going to try to enable the MMU, thank you.

closed 2020-05-21 by chregu82

chregu82, 2020-05-24:
Maybe a last hint for other uspi users. If you don't want to go through the process of setting up the MMU, there's a compiler option -mno-unaligned-access.
If you compile uspi with this option, possible unaligned accesses are eliminated by only reading one byte at a time.