rsta2/uspi #29: Cant seem to get this to work on my os

ajh123, 2020-10-14:
I have tried a few things and a few other drivers, this is [my OS](https://github.com/ajh123/pyOs/tree/in-dev)
Can you create a fork and help , On line 102 of main.c , I would like to read the keyboard into a buffer that clears on every newline (Enter key)

rsta2, 2020-10-15:
At first your OS must provide some functions to USPi, so that it can work. These functions are declared in the header file [uspios.h](https://github.com/rsta2/uspi/blob/master/include/uspios.h). I had a quick look and for example I'm not sure, if you have interrupt (IRQ) support, which is needed in any case. When all needed functions are provided, you can register a callback function, which receives the pressed keys using [USPiKeyboardRegisterKeyPressedHandler()](https://github.com/rsta2/uspi/blob/master/include/uspi.h#L55).

ajh123, 2020-10-15:
Thank you, i will try and gets IRQs working, then we can try from there

rsta2, 2020-10-16:
OK. You need all services declared in [uspios.h](https://github.com/rsta2/uspi/blob/master/include/uspios.h) (e.g. memory allocation and timer too).

rsta2, 2023-11-10:
There was no further reaction of the originator of this issue for a long time. Perhaps the issue has been solved without notice.

closed 2023-11-10 by rsta2