rsta2/uspi #30: Question about StartKernelTimer

sjtugjy, 2020-10-22:
Dear author,

Could you explain why it is necessary to invoke "StartKernelTimer"?
Specifically, the function is used in both lib/usbkeboard.c and lib/dwhcidevice.c.

Thanks in advance for considering this.

Jinyu

rsta2, 2020-10-23:
*StartKernelTimer()* is invoked to start a timer, which expires after the given amount of time and calls a callback function. This is used in *dwhcidevice.c* for implementing the USB interrupt transactions. On USB interrupt endpoints an transfer attempt is automatically made after an fixed interval, which is configured in the endpoint descriptor of the interrupt endpoint. Interrupt endpoints are used e.g. for HID devices like USB keyboards. There is an invocation of *StartKernelTimer()* in *usbkeyboard.c* too, which is used to implement a keyboard repeat function. But this is not activated by default.

sjtugjy, 2020-10-26:
Sorry for the late response. Thanks a lot.
I am porting uspi to a custom OS.
I have another question: TPropertyBuffer *pBuffer = (TPropertyBuffer *) MEM_COHERENT_REGION; 
The MEM_COHERENT_REGION seems to be an arbitrary address while it can be directly converted to bus_address.
Could you give me some instructions? Thanks.

rsta2, 2020-10-26:
MEM_COHERENT_REGION is a special region in memory, which is marked as "coherent" in the page table. This is needed for property mailbox access on the Raspberry Pi 3. Which instructions do you need?

sjtugjy, 2020-10-27:
The region must be fixed to 0x400000? Or, any physical memory address can be transformed with BUS_ADDRESS?
Btw, is there any reference manual on this to read? I did not find keywords like “property” in the following manual.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/QA7_rev3.4.pdf
https://cs140e.sergio.bz/docs/BCM2837-ARM-Peripherals.pdf

rsta2, 2020-10-27:
The coherent region can start at any address, which can be mapped in the page table. The USPi environment uses 1 MByte sections, so it starts at an 1 MB boundary and is 1 MB in size. Have a look at [this description](https://github.com/raspberrypi/firmware/wiki/Mailboxes) of the mailboxes. The property tags will be exchanged on channel 8.

sjtugjy, 2020-10-27:
Thanks! Now I get stuck for the first time to invoke "DWHCIDeviceTransferStage".
In this routine, the statement "while (pThis->m_bWaiting)" will never return.
According to the original code, it seems that "malloc" just returns normal memory.
I am not sure whether it is related to device/normal memory.

I am working on AArch64 with MMU enabled.

rsta2, 2020-10-27:
Hanging in this statement is mostly caused by an IRQ handler, which does not work. Are you sure, your IRQ is working? `malloc()` returns "normal" memory, but must be callable (as `free()`) from an IRQ handler.

sjtugjy, 2020-10-27:
At the beginning, IRQ is working (the timer irq handler can print).
However, when blocking in the initilization of uspi, it seems that IRQ no longer works (no timer irq output).

sjtugjy, 2020-10-27:
The problem is caused by accessing an fault memory address in the USB interrupt handler. I will check what happened.

sjtugjy, 2020-10-29:
The problem is caused by DWHCITransferStageDataGetDMAAddress which returns u32 as the address.
However, I am using AArch64, and the kernel runs at the high addresses. So, u64 should be used instead in such a case.

sjtugjy, 2020-10-29:
Thanks again, rsta2!

rsta2, 2020-10-29:
You are welcome. I suppose, you cannot use addresses above the 1 GByte boundary here, because the DMA address has to be converted into a bus address. I never tried USPi with a kernel running at high addresses.

sjtugjy, 2020-10-31:
Dear rsta2, I have to seek your help again.

I have successfully ported uspi to the kernel-mode (high addresses) of my OS (some efforts like page table mapping are needed).

Now, I try to move it into a user-mode process (also with some modifications). The main thread executes the initialization code and another thread executes the interrupt handler (the kernel will invoke such thread after receiving a USB interrupt).

Nevertheless, the interrupt handler reports an error for the second execution time.
The error is: dwhci: Transaction failed-1 (status 0xA).
For the first time, the dwhci transaction succeeds.

status 0xA seems to be 
#define DWHCI_HOST_CHAN_INT_HALTED                      (1 << 1)
#define DWHCI_HOST_CHAN_INT_STALL                       (1 << 3)

I have no idea now. Could you give me some suggestions?

sjtugjy, 2020-10-31:
"I suppose, you cannot use addresses above the 1 GByte boundary here, because the DMA address has to be converted into a bus address. I never tried USPi with a kernel running at high addresses."

Dear rsta2, I have to seek your help again.
Does malloc need to return the virtual address which is the same as the kernel address?

rsta2, 2020-10-31:
Normally yes. If you modify the BUS_ADDRESS() macro in *include/uspi/bcm2835.h*, perhaps it's possible to use a non-one-to-one mapping. But this has not been tested. In any case the resulting physical DMA address must be below the 1 GByte boundary. To convert it to a bus address the most significant two bits have to be set.

sjtugjy, 2020-10-31:
Thanks.
I am still confusing about why the timer is required for the USB interrupt handler.
From my understanding, when an interrupt comes, the handler is invoked (e.g. get the keyboard input and return).
Why the handler needs to set another routine which is invoked after some time.

rsta2, 2020-10-31:
You're welcome. This is one of the disadvantages of the USB host controller of the RPi 1-3. More modern or more integrated controller (e.g. xHCI) can poll the USB keyboard on its own, until a key is pressed. The host controller of the RPi 1-3 is not able to do this. So interrupt transfers have to be emulated using a timer. The actual transfer is regularly completed with a NAK or NYET packet and has to be restarted after an interval by the driver.

sjtugjy, 2020-10-31:
Currently, for porting uspi as a user-level driver, I create three threads:
- The main thread executes the initialization code.
- One thread executes the interrupt handler (the kernel will invoke such thread after receiving a USB interrupt).
- A third thread periodically wakes up (10ms) and fires the time-up callbacks.

Sometimes the user-level driver works. But sometimes it fails during the initialization phase.
Assertion fails, like:
expr: pThis != 0, file: ../drivers/uspi/lib/usbmouse.c, line: 81

Any idea?

rsta2, 2020-10-31:
Sounds reasonable. But the assertions are odd. `USBMouseDeviceConfigure()` is called [here](https://github.com/rsta2/uspi/blob/master/lib/usbdevice.c#L397) and only if the handed over pointer is not zero. But the first assertion claims, the pointer is zero. How can this happen?

sjtugjy, 2020-11-09:
Hi rsta2,

I solved the problem (a bug in my kernel) but forgot to close this issue.
Thanks for your patience!

--
Sincerely,
Jinyu

closed 2020-11-09 by sjtugjy