rsta2/uspi #31: Support for EndpointTypeIsochronous

wzx-ipads, 2020-11-10:
I'm trying to implement a uvc driver based on USPi. However I found isochronous endpoints used for data transferring seem to be unsupported. I made attempts to solve this problem but failed. Could you please offer me some suggestions? Thanks a lot!

rsta2, 2020-11-10:
Unfortunately there is no isochronous support in USPi and adding it seems to be not possible to me. When I analyzed the Linux driver code of the USB driver years ago, I left the isochronous support out to reduce complexity. You would have to analyze this driver again. If it would be simple to add this support, I would have done it.

wzx-ipads, 2020-11-10:
Thank you! Hope I can find a way to work around this problem.

closed 2021-01-08 by rsta2