rsta2/uspi #32: Keyboard issue

chregu82, 2021-01-16:
Hello
I ran into a problem if I hold down a key for at least 30 seconds.
I'm using a RAW handler which gets registered with USPiKeyboardRegisterKeyStatusHandlerRaw(). Usually this works fine and my handler gets called on every key press and key release.
However if I hold down a key for a long period of time, at least 30 seconds, the key released event is missing and from that point on, no more key events are reported.
I'm using your develop branch.

rsta2, 2021-01-17:
Unfortunately I cannot reproduce this problem. I tested this with two different keyboards. After holding a key for 40 seconds, both keyboards still respond. Maybe your keyboard switches off, after holding a key for that long, which is not usual?

chregu82, 2021-01-17:
Thank you for your help. I just tried different Pi's. It seems to work fine on my Pi Zero V1.2, but stops working on my Pi2 and Pi3. What hardware did you test with?

rsta2, 2023-11-10:
There was no further reaction of the originator of this issue for a long time. The issue will be closed, because it cannot be solved without interaction.

closed 2023-11-10 by rsta2