rsta2/uspi #33: Newbie question about porting

lonnietc, 2022-07-26:
Hello All,

I am working on a bare metal project for the x86_64 platform and am wondering if this library could be ported to that platform as well since it seems to have all of the functionality that I need, but is currently on for my platform.

Any thoughts for suggestions would be greatly appreciated.
Thanks in advance

rsta2, 2022-07-29:
Are you sure, your target platform uses a DWC OTG USB controller, which is the only supported controller by USPi? Otherwise this library won't be of much use for you.

rsta2, 2023-11-10:
This issue is not really USPi related and there was no further reaction from its originator for over a year. I think, it can be closed.

closed 2023-11-10 by rsta2