rsta2/uspi #35: Sample keyboard not initializing

Java6502, 2023-10-23:
Hello,
I am trying to run one of the included samples (keyboard) on a raspberry pi 2 model b.  For some reason, the keyboard sample does not work.  The best I can figure so far is that the problem occurs during the call to USPiEnvInitialize().  I did some investigating I tracked down the source of the problem to the function TimerInitialize in timer.c
Everything works until

        write32 (ARM_SYSTIMER_C3, read32 (ARM_SYSTIMER_CLO) + CLOCKHZ / HZ);
	
	TimerTuneMsDelay (pThis);

	DataMemBarrier ()

I know that timing is very critical for USB, so I am wondering if I have configured something incorrectly specifically for the RPI2B.
Anyway, I have no problem getting the similar example (08-usbkeyboard) from the circle library to work.  I looked at the timer.c file from circle and it's a bit different from the timer.c in uspi.  Any ideas as to what I am doing wrong. 

Thanks so much,
Jonathan

rsta2, 2023-10-25:
Hi Jonathan, I tried the keyboard sample here again and it worked. I used the GNU-C-based toolchain 12.2.Rel1 from [this website](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) and the following firmware:

```
$ strings start.elf | fgrep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 10:52:42
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Mar 17 2023
VC_BUILD_ID_BRANCH: bcm2711_2
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean)
```

The file *bcm2710-rpi-2-b.dtb* is not necessary on the SD card. Your setup must be different in some way, otherwise it should work. Do you have the RPi 2B second edition, which uses the BCM2837 SoC of the RPi 3 by any chance? I have the first edition. I don't think, that the keyboard is the problem, because the problem arises, when the keyboard has not been detected yet.

Rene

rsta2, 2023-11-03:
Are there any news on this?

Java6502, 2023-11-05:
I have a RPI2B v1.1 and v1.2  No problem running this on the RPI2B v1.1, however the example does not work on the RPI2B v1.2
I am still trying to track down what the problem is, but no luck so far.  There is another project that uses uspi (PigFX).  It's a simple terminal emulator.  I has its own environment, and works just fine on the RPI2B v1.2  This leads me to think the underlying problem must be in the environment and not the uspi library itself.

rsta2, 2023-11-05:
Please build with `RASPPI=3` set in *Config.mk* for the RPi 2B v1.2. This 
should work. This RPi version uses the BCM2837 SoC of the RPi 3, so this 
setting is required here. The kernel image is named *kernel8-32.img*. In 
Circle additional work has been done, so that `RASPPI=2` works for the RPi 2B 
v1.2 too. This is not the case in the USPi environment.

Java6502, 2023-11-06:
That worked.  Thanks for the help

closed 2023-11-06 by Java6502