rsta2/uspi #37: Does uspi support qemu?

endinferno, 2023-12-16:
In the overview, it is said that "USPi was "mechanically" ported from the Circle USB library which is written in C++. That's why the source code may look a little bit strange. But it was faster to do so."

As far as I know, circle supports Qemu. So does uspi support qemu as well?

rsta2, 2023-12-16:
No, USPi and the USPi Environment does not support QEMU. There have been many 
modifications applied to Circle, after its USB driver had been ported for USPi. 
Some of these modifications are necessary for QEMU support and are not 
available in USPi.

endinferno, 2023-12-17:
I'm trying to get uspi to support qemu, although I don't know how long it will take. Will this work take a long time?
And I'll keep you in touch as soon as I make any progress.

rsta2, 2023-12-17:
OK. I don't think there are many changes necessary. The good thing is, that 
you can debug it using GDB. What I have seen in a quick debug is, that you 
have to remove the following block for QEMU:

https://github.com/rsta2/uspi/blob/master/lib/dwhcidevice.c#L907-L920

Also the firmware mailbox call, used to create the frame buffer, seems to be not 
functional with QEMU:

https://github.com/rsta2/uspi/blob/master/env/lib/bcmframebuffer.c#L115

You have to port the property mailbox calls, which are used by Circle here.

endinferno, 2023-12-18:
Thanks for your help!

That's an important hint for me. I remove the code in https://github.com/rsta2/uspi/blob/master/lib/dwhcidevice.c#L907-L920 as you told me. Now it can pass the function `USPiInitialize`.

It can not detect the device, I print the device names in the QEMU which are "ven409-55aa, dev9-0-0", "ven525-a4a2, dev2-0-0".
On my raspi3B+ board, I also have another device name "ven424-7800" which is a lan7800 network device.
So I think there still exists some problems.

As for mailbox, I find that in my code which has been updated for a little, the mailbox operation is little including `SetPowerOn` and `GetMacAddress` and QEMU logs correctly. So I think maybe there will be no problem for MailBox?

And I launch QEMU with `-netdev user,id=net0 -device usb-net,netdev=net0`, it is the same as the script which I used for testing the sample of Circle. So I think the script is correct.

rsta2, 2023-12-18:
QEMU does not support the LAN7800 device (ven424-7800). For network access it 
uses a USB CDC Ethernet device (int2-6-0), which is not supported by USPi, but 
by Circle (lib/usb/usbcdcethernet.cpp). If you need network access with USPi 
in QEMU, you have to port it.